schnoog/Joystick_ESP32S2 #2: Joystick not recognized

hfjg, 2023-09-30:
#Issue
Examples "FlightControllerTest.ino" and "JoystickTest.ino" both do not report HID device. Actually, nothing is reported.
Test with flashing the onboard LED works, so I expect the program is transferred correctly.

## Hardware
S2_DevBoard-clone  by "VCC GND", separate usb connectors for native usb and UART

##PC
Linux Mint (feat. Ubuntu 20.3), tested with lsusb, dmesg and syslog, no USB from board reported

schnoog, 2023-09-30:
Hi,
I tries the examples on a S2 and a S3 board. Before i pushed them to github.
Using the Arduino IDE it's necessary to 
"enable USB CDC on boot".
Additionally I have chosen the USB-Mode USB-OTG(TinyUSB)

Unfortunately I'm currently not aware how to set this in Platfromio.

hfjg, 2023-09-30:
Thank you. I'll check tomorrow.

schnoog, 2023-09-30:
You're welcome.

`build_flags = 
	-DARDUINO_USB_CDC_ON_BOOT=1`

in your platformio.ini should do the trick.

schnoog, 2023-10-02:
Great to hear that.
Thank you very much for your feedback

closed 2023-10-02 by schnoog