schnoog/Joystick_ESP32S2 #4: Could be cool to add Force Feedback in the future

tiking, 2023-10-07:
Very cool project, I'll try it for sure :)

It could be cool to add force feedback in the future like on this library :
https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

Sébastien,

schnoog, 2023-10-08:
Hi and thanks

can not promise that it will become reality, but I can promise that I'll have a look into it

serg-bloim, 2023-10-23:
That would be awesome!

schnoog, 2023-10-24:
Hi, thanks for your feedback.
I had a look into it and decided to give it a try (not what is a called low hanging fruit)
Can't promise any time frame

serg-bloim, 2023-10-25:
> Hi, thanks for your feedback. I had a look into it and decided to give it a try (not what is a called low hanging fruit) Can't promise any time frame

That's super cool. I'm currently building a steering wheel using your lib. Will be watching for updates.

schnoog, 2023-12-17:
This thing drives me crazy....
After investing "some" hours, I'm still not able to get it working. 
Hopefully I will find some time to dive in deeper into the topic (USB specs aren't the biggest pleasure to read) during my XMAS break.

gagagu, 2024-06-26:
Hi,
I have made an FFB Yoke ([www.gagagu.de](https://github.com/gagagu/Arduino_FFB_Yoke/)) based on the GitHub project "FINO" (FFB Wheel) , which also based on the project shown above. FINO combined the project your project based on with the FFB library and I have made a FFB Yoke with it.

So the Ardunio is great but using an ESP32 would be much better because of speed and RAM. I am working on an implementation for esp32-s3. After testing you sources i will implement step by step the ffb code but I am not so good in HID programming. Its not easy to get into. I have made some code inspection an i guess all methods FINO is used are also available in the library path (AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\libraries\USB) so i guess it can be a success.

May be we can combine our foces to get an FFB library working for ESP32?
You can contact me on my Discord https://discord.gg/uuw7TWVu