schnoog/Joystick_ESP32S2 #5: Allowed Values

ksiezopolski, 2023-11-08:
Hey
It would be great to know possible maximum and minimum values. Defaults are great but to customise we need to know available ranges. For example, the default number of buttons is 32. Can I set it to 0 or 10000?
The original library was restricted to 6 axes, is this the same case in this library for ESP32? Or can I use let's say 11 axis?
Can I use all 16-bit analogues and use them all at once with 1024 buttons using a shift register?
Please provide a complete datasheet for this library with minimum and maximum values on top of pre-set defaults.
Kind Regards

schnoog, 2023-11-09:
Hi, thanks for your comment.

The axis range can be the full int32_t range, from -2,147,483,648 to 2,147,483,647.
However, I would suggest to not exceed the range of -32’767 to 32’767 since the Windows function IDirectInputDevice8::GetDeviceData, which processes the device data, MAY only be able to cover this range (unfortunately I don't have the Win-documentation at hands to check it.
Windows will then use the center-value between upper and lower limit as center point for the axis.

In regards to the number of axis:
To my knowledge "Direct Input" only allows up to 8 joystick axis
(X,Y,Z, Rx, Ry, Rz as well as trottle/accelearot and brake) with up to 128 buttons. 
So you may be able to add more, but there's no guarantee that he joystick will be recognized by your OS afterwards.

If you need more buttons / axis, you can defined multiple joysticks to spawn in your code. 
(my first sandbox test was with 4 joysticks, each 4 axis and 128 buttons.

ksiezopolski, 2023-11-09:
Dear Schnoog

Thank You for your response. I was able to get 1024 buttons using Pro-Micro
with shift registers in the matrix, works great as push buttons but had no
way to make it work properly with toggle switches (Not enough of mem.). I
was hoping to revisit my ideas using way more powerful ESP32 S3. I would
like to make boards for flight simulator fans with at least 8 analog axes,
256 toggle switches, and at least 6 rotary encoders. In theory, it should
be possible, but the previous joystick library was limited to 6 analog
devices and 128 buttons in total. It was also not enough pins to
accommodate rotary encoders.
I will follow your article and set new ranges to see how esp32 will handle
my setup.

Once again Thank You

Michael Ksiezopolski

P.S. Please check my PCBs on eBay under 2013ukksiez.