schnoog/Joystick_ESP32S2 #7: How to flash onto an ESP32S3 Dev Kit C N8R2

rubdul, 2023-12-31:
Hello I'm having trouble getting this library to work with an esp 32 S3 dev kit C. My unit has two usb type c ports as depicted below.
[image]
My problem is that when I make a sketch in arduino ide and attempt to upload it to the board, the esp32 is still not picked up by my computer as a USB HID device. The best I have managed is getting the be detected by window as something different than the default ESP32-S3 it would show a message saying that Windows does not recognize the USB device. I have tried flashing it with the OTG and the other port with different device settings and tools enabled but haven't had much success. I also performed the enable usb .How would one get this to work for esp32-S3 Dev Kit boards? Which settings and IDE tools would I need to enable. Here is a link to the original board and the protoype sketch. https://www.amazon.com/gp/product/B0BPP28ZFB/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

`
#include <Joystick_ESP32S2.h>

using namespace std;

// Analog outputs
#define pitch 0 // Rx input
#define throttle 1 //  throttle input 

// Button and switches
#define button 4
#define pos_3_up 5
#define pos_3_down 6
#define mom_up 7
#define mom_down 15

// Joystick_(uint8_t hidReportId = JOYSTICK_DEFAULT_REPORT_ID, uint8_t joystickType = JOYSTICK_TYPE_JOYSTICK,
// uint8_t buttonCount = JOYSTICK_DEFAULT_BUTTON_COUNT, uint8_t hatSwitchCount = JOYSTICK_DEFAULT_HATSWITCH_COUNT,
// bool includeXAxis = true, bool includeYAxis = true, bool includeZAxis = true, bool includeRxAxis = true, bool
// includeRyAxis = true, bool includeRzAxis = true, bool includeRudder = true, bool includeThrottle = true, 
//bool includeAccelerator = true, bool includeBrake = true, bool includeSteering = true)

// We are going to include the Rx and Ry inputs dfo 

Joystick_ Joystick(0x11, JOYSTICK_TYPE_JOYSTICK, 5, 0, false, false, false, true, false, false, false, true, false, false, false ); // create joystick object

const bool initAutoSendState =true;

// set the axis on the joystick side
int RxAxis = 0;
int throttleAxis = 0;

//declare buttons
bool button1State = 0;
bool button2State = 0; // 3pos switch
bool button3State = 0; // 3 post down
bool button4State = 0; // mom down
bool button5State = 0; // mom up

void setup() {
  // put your setup code here, to run once:
  // First get the serial boot loader working
  Serial.begin(115200);       // begin serial monitor to see what is going out
  
  // Setup the button pins
  // pinMode(pitch, INPUT);  // collective angle input 
  // pinMode(throttle, INPUT);  // twist throttle 
  pinMode(button, INPUT_PULLUP); // button input
  pinMode(pos_3_up, INPUT_PULLUP); // 3 position switch
  pinMode(pos_3_down, INPUT_PULLUP); // 3 position switch
  pinMode(mom_up,  INPUT_PULLUP);// momentary up
  pinMode(mom_down, INPUT_PULLUP);// momentary down  

  // begin the joystick
  // duhhh shame to myself
  Joystick.begin();

  // Setup USB Names
  USB.PID(0x8211);
	USB.VID(0x303b);
	USB.productName("ESP32 S2 Collective");
	USB.manufacturerName("My company name");
	USB.begin();
  // build_flags =
	// ....
  //   // -DARDUINO_USB_CDC_ON_BOOT=1
  //   // // -DUSB_VID=0xF011
  //   // // -DUSB_PID=0xF011
  //   -DUSB_PRODUCT="Helicopter Collective"
  //   -DUSB_MANUFACTURER="Espressif Systems"

  

}

void loop(){ 
// lets do the axis inputs

// Axis runtime for pitch and throttle
RxAxis = analogRead(pitch);
RxAxis = map(RxAxis, 0, 1023, 0, 255);
Joystick.setRxAxis(RxAxis);

throttleAxis = analogRead(throttle);
// throttleAxis = 1023 -throttleAxis;
throttleAxis = map(throttleAxis, 0, 1023, 0, 255); // I reversed the axis
Joystick.setThrottle(throttleAxis);
Serial.print("throttle setting: ");
Serial.print(analogRead(throttleAxis));

// now include the digital buttons 
// NOTE: no pullup resistors used yet

// normal switch button code
bool currentButton1State = !digitalRead(button);
if( currentButton1State !=  button1State){
  Joystick.setButton(0, currentButton1State); // update joystick button0
  button1State = currentButton1State; // update the button variable
}
Serial.print(", ButtonState: ");
Serial.print(currentButton1State);

// 3position switch code
bool currentButton2State = !digitalRead(pos_3_up);
if( currentButton2State != button2State){
  Joystick.setButton(1,currentButton2State);
  button2State = currentButton2State;
}
Serial.print(", 3 position UP state:");
Serial.print(currentButton2State);

bool currentButton3State = !digitalRead(pos_3_down);
if (currentButton3State != button3State) {  
  Joystick.setButton(2, currentButton3State);
  button3State = currentButton3State;
    } 
Serial.print(digitalRead(pos_3_down));

// momentary switch code
bool currentButton4State = !digitalRead(mom_up);
if (currentButton4State != button4State) {
  Joystick.setButton(3, currentButton4State);
  button4State = currentButton4State;
}
Serial.print(", Momentary UP state:");
Serial.print(currentButton4State);

bool currentButton5State = !digitalRead(mom_down);
if (currentButton5State != button5State) {
  Joystick.setButton(4, currentButton5State);
  button5State = currentButton5State;
}
Serial.print(", Momentary Down state:");
Serial.print(currentButton5State);
Serial.println();
// Debounce time to prevent flooding the USB input
delay(10); // in milliseconds

}`

schnoog, 2024-01-01:
Hi and happy new year.

The Lolin S3 I use also come with 2 USB ports. The USB&OTG one is the one directly hooked up to the ESP chip and which is able to be defined. So be sure to use that one since the HID device will only be available at this port. 

I have seen some really strange behaviors with the USB, if you connect it and open the Windows USB-Game-controls ( [Windows-Button] + tying in USB should show it as one of the first results (depending on your language settings), is the joystick visible in it.

I copied your code into the Arduino IDE, used the following settings and compiled it.

[image]

After flashing it on the ESP32S3 and unplugging it (not only reset) and reattaching it, I see the following device in the Windows USB game controller settings

[image]

rubdul, 2024-01-03:
Happy New Year. Thanks for taking the time to reply. I'm still getting the windows error even with these settings. Are you flashing as the ESP32S3 Dev Module or as the ESP-32-S3-USB-OTG.

schnoog, 2024-01-03:
I'm only using the USB-OTG port for such things.  
Hi, USB stack is a holy mess. Windows tries to remember devices which makes the thing even more tricky.

Just checked 2 other projects which are working at my nephews computer.
For both I initiated the joystick  with
Joystick_ Joystick(0x03
while your code uses
Joystick_ Joystick(0x11

No idea if that's the problem, but please give it a try.

rubdul, 2024-01-12:
I tried that but it didn't work. Then I decided to re-enable erase all flash before sketch upload, and reuse the uart usb port for flash. A little more fiddling later and it worked. Thank you so much friend I am blessed to be able to talk to people like you in this community.

closed 2024-01-13 by schnoog