schnoog/Joystick_ESP32S2 #8: Question regarding using esp32 s3 with analog read 12bit resolution

mittlc, 2024-01-01:
Hello there,
i am experiencing some strange behavior when using the 12 bit analog read of a esp32 s3 with this library to create a USB Gamecontroller (two joystick, each providing x/y axis which will be mapped to x,y,z and throttle.)

After calibration (which seems to work fine) for example the x/y axis will move only in two direction from the center. So to the left and up, but not to the right and down - although i can see on my transmitter serial monitor that values are sent correctly.

This issues does not occur if i map my readings to a 10 bit resolution 0 - 1024 with 512 beeing the center - but then i am loosing the resolution of the esp32, which is a pitty.

Any idea on why that might be?

[image]
[image]
[image]
[image]

schnoog, 2024-01-01:
Hi and happy new year,
without seeing your code it's hard to say what's going wrong. 

I used this (quick and dirty) example and just tried it, it works.
Did you set the axis range in the setup?

'''#include <Joystick_ESP32S2.h>

Joystick_ Joystick(0x11, JOYSTICK_TYPE_JOYSTICK, 5, 0, true, true, false, true, false, false, false, true, false, false, false ); // create joystick object

const bool initAutoSendState =false;

int direction = 0;
int x = 0;
int y = 0;
void setup() {
Serial.begin(115200); // begin serial monitor to see what is going out
Joystick.begin(false);
Joystick.setYAxisRange(0,32767);
Joystick.setXAxisRange(0,32767);

// Setup USB Names
USB.PID(0x8211);
USB.VID(0x303b);
USB.productName("ESP32 S2 Collective");
USB.manufacturerName("My company name");
USB.begin();
// build_flags =
// ....
// // -DARDUINO_USB_CDC_ON_BOOT=1
// // // -DUSB_VID=0xF011
// // // -DUSB_PID=0xF011
// -DUSB_PRODUCT="Helicopter Collective"
// -DUSB_MANUFACTURER="Espressif Systems"

}

void loop(){
  if (x < 1000){
    direction = 1;
  }
  if ( x > 31765){
    direction = 0;
  }

  if (direction == 0){
        x = x - 500;
        y = y - 500;
  }else{
        x = x + 500;
        y = y + 500;
  }
  Joystick.setYAxis(y);
  Joystick.setXAxis(x);
  Joystick.sendState();
  delay(20);

}'''

mittlc, 2024-01-01:
Hello schnoog,
thanks a lot and happy new year to you! I think setting the range is what i was missing - now it is workingas expected.

closed 2024-01-01 by mittlc