schnoog/Joystick_ESP32S2 #12: Simultaneous serial and gamepad data

ChrGri, 2024-04-20:
Thank you for this awesome project. 

I'm using an ESP32 S3 (connected via USB port) and want to send/read additional serial data over the USB port to monitor/debug the ESP.

What seems to work:
- USB gamepad data
- ESP seems to receive serial data from PC.

What does not work:
- ESP doesn't send serial data to PC.

Do you have an example script where the additional serial communication is working properly?

BR
Chris

ChrGri, 2024-04-20:
I wrote a simple test script (see attachment) and it does what I'm expecting: changing the USB gamepad output and printing info to the Arduino serial monitor simultaneously.
[Joystick_Serial_test.zip](https://github.com/schnoog/Joystick_ESP32S2/files/15048854/Joystick_Serial_test.zip)

My original code has two tasks running on two cores which are accessing the same USB port. Apperently that creates some trouble. I'll continoue the serach for a solution.

schnoog, 2024-04-20:
Hi Chris, thanks for your feedback.

In my projects I never used the serial output and both cores at the same time. 
Thanks for having a look into it. Would appreciate if you could keep me updated.

Cheers
Volker

ChrGri, 2024-04-21:
Can you confirm that the flash settings are okay?

[image]

The USB cable is connected to the USB OTG port.

BR
Christopher

schnoog, 2024-04-21:
What I'm seeing on your screenshot looks correct to me.
(don't have ArduinoIDE installed at the moment, so assessment based on my memory) 

A news delivery of S3 should arrive soon, will start testing once they arrived. My current ones are all wired to my simulator setup.

ChrGri, 2024-04-27:
Hi @schnoog, 

the current status is, that when serial and HID output are asked at high repetition rate, both outputs stop working. I've addded a testscript below. Feel free to play with the delay.

[Joystick_Serial_test.ino.txt](https://github.com/schnoog/Joystick_ESP32S2/files/15139607/Joystick_Serial_test.ino.txt)

BR
Chris

kbastronomics, 2025-01-15:
Is there any update on this issue,  I'm having same issue i think
I'm using this with DCS-BIOS that sends data to my device and I use the Joystick device to send button/axis information to DCS-BIOS.

it works briefly once started up,  then i loose the HID devices but the serial devices keep working.  it's like HID got turned off
it seem to happen with a DTR signal like the OP reported.

kbastronomics, 2025-01-15:
I take that back,  DTR does not cause it in my case.  it just took longer before HID froze

schnoog, 2025-01-15:
@ChrGri opened a ticket for that issue https://github.com/espressif/arduino-esp32/issues/10307