schnoog/Joystick_ESP32S2 #14: can i use a rotary encoder with the code??

eydoooo, 2024-07-09:
im trying to make a steering wheel i have used  apot but the 270 degreees isnt realy good tbh so i was wondering if i could use a rotary encoder with the code or using a hdd motor/stepper as a rotary encoder

schnoog, 2024-07-09:
Hi, for sure you can use a rotary encoder for this.
Create a global variable (f.e. 0 to 4095) and increase or decrease the value with each step based on the direction of the rotation.

However, I doubt that you will be happy with a RE as steering wheel.
Even if you use one with 24 steps per rotation that means you have 15° steps.
So a deadzone of 15° at each of the steps. 
That's like driving a old rusty car with worn out control arms. 

Just some ideas:
- Multiturn potentiometer (as the name states, multipl turns for full deflection)
- Hall sensor f.e. AS5600 giving you 360°
- Hall sensor with a simple belt or gearwheel with a f.e. 1:2 transmission giving you 720°

eydoooo, 2024-07-09:
> Hi, for sure you can use a rotary encoder for this. Create a global variable (f.e. 0 to 4095) and increase or decrease the value with each step based on the direction of the rotation.
>
> However, I [...]

Thank you for your reply. To be honest I actually don't have a rotary encoder to make a steering wheel, I just thought to ask about it since what I had in mind was to use a hdd motor or a stepper motor as a rotary encoder. And also since they both use the same library. I'm just surviving on what I have since my country is literally going through an economy crisis😅. I'm actually interested in the multi turn pot since it's the cheapest of all, and I'm pretty familiar with pots. Unlike hall sensors.

closed 2024-07-09 by schnoog