schnoog/Joystick_ESP32S2 #16: HID unresponsive

YFFKJB, 2024-12-05:
In the current program, the computer can correctly identify the HID device, but when the computer is restarted after sleeping, although the device can be identified by the computer, the data of the axis and keys on it are no longer updated. What is the reason?

schnoog, 2024-12-05:
Hi, 
I use the library on an ESP32S3 for my flight-throttle for one and a half year and didn't had that problem yet.
How does your code look like?

YFFKJB, 2024-12-06:
#include <Arduino.h>
#include <SimpleFOC.h>

#include <Joystick_ESP32S2.h>

#define BUTTONPIN 38

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  1, 0,                  // Button Count, Hat Switch Count
  true, false, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

void setup() {

  USB.PID(0x9884);
  USB.VID(0x031F);
  USB.productName("JOY");
  USB.manufacturerName("MY_JOY");
  USB.begin();
  
  Joystick.setXAxisRange(-32767,32767);
  Joystick.begin(false);

}

void loop() {
  
  int currentButtonState = !digitalRead(BUTTONPIN);
  if (currentButtonState != previousButton1State) {
    Joystick.setButton(0, currentButtonState);
    previousButton1State = currentButtonState;
  }

  Joystick.sendState();

}

I used the library files of SIMPLEFOC and rocker _ESP32S2. In the code, I enabled an axis and a key. Recently, it has been found that turning on the computer after each hibernation can identify the devices on the computer, but it does not respond to the key and axis data. In this way, the operation will suddenly fail in the game. Every time I have to press the reset button on ESP32 to restart, this problem can be solved. I use ESP32S3, thank you for your reply.

schnoog, 2024-12-09:
The code looks good to me and I wouldn't expect any issue with it. 
However, different USB ports behave differently. Some keep the power on, some even the data line, same for different USB hubs.

Please check your device manager and 
-enable the "allow the computer to turn off this device...."
-disable the "allow this device to wake the computer"
options for your joystick.

If that's already the case you could try a different USB port.