schnoog/Joystick_ESP32S2 #17: Default input pins

redo33, 2024-12-09:
Thank you very much for porting the library to ESP32 S2/3. :)  

I tried the JoystickTest.ino, and my Wemos S2 Mini can be detected as a TinyUSB HID device on Windows.  However, none of S2's input pins is not detected.  The Windows Joystick properties shows no activity at all.

I wonder how physical input pins are assigned by the Joystick_ESP32.  From [Dan's tutorial video](https://youtu.be/hoCOq9Ngp44), it seems

x-axis y z rudder throttle  = Pin A0 A1 .... 
Btn 1 starting from Pin 9 .... 

I'm not sure whether the input pins are automatic enumerated.  

For analog inputs, it starts with Pin A0 ?

Flightcontrollertest.ino

```
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 32, 0,
  true, true, false, false, false, false,
  true, true, false, false, false);
```
11 axes in total.

I tried the axis input but it didn't work too.

Any advice would be greatly appreciated. 

Thank you very much.

schnoog, 2024-12-09:
Hi, the code you posted does only initiate the joystick but does not send any "commands" since there's not automatically enumeration.

Here's a small snippet

In Setup:
    Joystick.setXAxisRange( 1023, 0);
    Joystick.setYAxisRange(0, 1023);

In Loop:
        int AX =  analogRead(0);
        int AY = analogRead(1);
        Joystick.setXAxis(AX);
        Joystick.setYAxis(AY);
        ...
        ...
        Joystick.sendState();

Maybe this old sketch (uses the Arduino Library but is code compatible beside of the library inclusion) is helpful to you 

https://github.com/schnoog/joystick_F16_sidestick/blob/34b556b609e60c4defac36893d9e67f40735f712/F16Sidestick_Arduino.ino

schnoog, 2024-12-09:
Great to hear that. Wish you a lot of fun with it

closed 2024-12-09 by schnoog