schnoog/Joystick_ESP32S2 #22: Works somewhat but is sluggish and has some errors

musseilm, 2025-03-14:
Tried to basic examples on a esp32s3 but getting the following errors in the serial monitor:

[E][USBHID.cpp:359] SendReport(): report 3 wait failed

Buttons seems sluggish to respond if at all. Joystick axis seems to work ok but slow.

schnoog, 2025-03-14:
Hi,
using serial output in combination with the USB HID unfortunately can cause some problems

https://github.com/schnoog/Joystick_ESP32S2/issues/12

An upstream issue was opened but not resolved (yet) and there's nothing I can do to change this behavior.

https://github.com/espressif/arduino-esp32/issues/10307

To have the joystick working flawless you need to remove the serial output

musseilm, 2025-03-14:
Even with serial removed buttons seem to not work

schnoog, 2025-03-16:
Can you please post your code (as well as the settings you made in the ArduionIDE / Platformio.ini)