schnoog/Joystick_ESP32S2 #24: Only seven axes show in usb controllers

dsaniel, 2025-05-11:
I built a joystick using an ESP32S3 and included 8 axes (XAxis, YAxis, ZAxis, RxAxis, RyAxis, RzAxis, Rudder and Throttle), when I enable debugging through serial monitor, I can see my joystick reporting changes for all axes, but, windows won´t recognize my rudder axis. In fact I tried a lot of different combinations and of all the "simulation" axes, windows will only recognize the rudder one.

Any ideas?

schnoog, 2025-05-12:
Hi,

did you check the rudder axis with the windows built in USB-Gamepad tool or with a game /different tool?

The Windows tool has some limitations which will not show many buttons, maybe that also is the case for the axis.

Could you please try it with a tool like https://www.planetpointy.co.uk/joystick-test-application/

Looking forward to your response

KR

dsaniel, 2025-05-12:
I tested with windows usb controller tool (the application you linked does not start in my system, W11, some System.InvalidOperationException).

If I remove any of the axis it is immediately reflected, but, for simulation axis the only one that I can see in the game controllers screen is the one for rudder.

Also, I tried mapping the axis in DCS in case it was a limitation of the windows game controller application, but DCS is not recognizing that axis either.

[image]

Here is the hid descriptor in IO Ninja.
[image]

dsaniel, 2025-05-12:
In case it helps, if I try the following test sketch with my ESP32S3 Dev, it only displays the same 7 axes.

`#include <Joystick_ESP32S2.h>

Joystick_ Joystick;

void setup() {
  USB.begin();

  Joystick.setXAxisRange(0, 4095);
  Joystick.setYAxisRange(0, 4095);
  Joystick.setZAxisRange(0, 4095);
  Joystick.setRxAxisRange(0, 4095);
  Joystick.setRyAxisRange(0, 4095);
  Joystick.setRzAxisRange(0, 4095);
  Joystick.setRudderRange(0, 4095);
  Joystick.setThrottleRange(0, 4095);  
  Joystick.setAcceleratorRange(0, 4095);

  Joystick.begin();
}

void loop() {
  Joystick.setXAxis(200);
  Joystick.setYAxis(600);
  Joystick.setZAxis(1000);
  Joystick.setRxAxis(1400);
  Joystick.setRyAxis(1800);
  Joystick.setRzAxis(2200);
  Joystick.setRudder(2600);
  Joystick.setThrottle(3000);
  Joystick.setAccelerator(4000);
  
  delay(50);
}`

dsaniel, 2025-05-12:
And finally, I see changes in the missing axis reported in IO Ninja, but neither windows, nor DCS seem to get them.

[image]

schnoog, 2025-05-12:
It's only a guess, but Windows has a strange behavior if a HID device changes after it was seen by the system the first time.

Try to change VID or PID  (I usually use the VID 8798, but during development 8799).
Alternatively there's a method to remove old entries from registry, but I don't remember exactly how (Windows is only started for DCS, daily driver is linux).

dsaniel, 2025-05-12:
I actually tried the joystick in a fresh windows and got the same thing, rudder is not showing.

schnoog, 2025-05-13:
Can you please share your complete code. Will try to identify the issue

dsaniel, 2025-05-14:
OK, I checked with linux and it does show the 8 axes, the only thing is that is shows throttle as 7 and rudder as 8.

[image]

dsaniel, 2025-05-14:
More information, I checked my WW UFC, that is the only "joystick" I have with more than 7 axes and they seem to be using slider and dial, instead of simulation axes. This seems consistent with the comments in [this](https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/221) thread.

[image]

I'm not very familiar with HID Descriptors, how hard would it be to add slider and dial support to the library? If you are not very familiar either or you don't have the time, I'll try to figure it out myself.

dsaniel, 2025-05-14:
Wow, that was fast, thank you, I’ll try to test it when I get back home!

Thanks a lot!

dsaniel, 2025-05-15:
OK, I tried it and it's not working for me. With the code I'm pasting below I get the following error in Serial Monitor: 
[292900][E][USBHID.cpp:362] SendReport(): report 3 wait failed

In case it helps, this is what the descriptor looks like:
[image]

This is what windows "sees":
[image]

And this is the code:
`#include <Joystick_ESP32S2.h>

Joystick_ Joystick = Joystick_(
        JOYSTICK_DEFAULT_REPORT_ID,
        JOYSTICK_TYPE_MULTI_AXIS,
        0,
        0,
        true,
        true,
        true,
        true,
        true,
        true,
	      true,
	      false,
        false,
        false,
        false,
        false);

void setup() {
  USB.begin();

  Joystick.setXAxisRange(0, 4095);
  Joystick.setYAxisRange(0, 4095);
  Joystick.setZAxisRange(0, 4095);
  Joystick.setRxAxisRange(0, 4095);
  Joystick.setRyAxisRange(0, 4095);
  Joystick.setRzAxisRange(0, 4095);
  Joystick.setSliderRange(0, 4095);
  Joystick.setDialRange(0, 4095);
  // Joystick.setRudderRange(0, 4095);
  // Joystick.setThrottleRange(0, 4095);  
  // Joystick.setAcceleratorRange(0, 4095);

  Joystick.begin();
}

void loop() {
  Joystick.setXAxis(200);
  Joystick.setYAxis(600);
  Joystick.setZAxis(1000);
  Joystick.setRxAxis(1400);
  Joystick.setRyAxis(1800);
  Joystick.setRzAxis(2200);
  Joystick.setSlider(3400);
  Joystick.setDial(3800);
  // Joystick.setRudder(2600);
  // Joystick.setThrottle(3000);
  // Joystick.setAccelerator(4000);
  
	delay(50);
}`

BTW, I'm happy to go into a remote online debugging session if you think that could help.

dsaniel, 2025-05-16:
OK, this is very weird, I finally decided to take a look at the hid descriptor spec and checked the raw descriptor for my button box, for some WEIRD reason windows reports (or at least IO Ninja does) a HID descriptor that is nothing like what your code should be generating:

[image]

So I checked in Mac it actually reports the hid descriptor that I would expect from your code...

[image]

So, at this point, I guess it's windows being windows. I'll try to play around with device id to see if it is some kind of left over in the registry, although a quick search of the internet says that windows does not store HID descriptors anyway and that it reads them when you connect a device...

dsaniel, 2025-05-18:
OK, I managed to get it working for my case (8 axes), this works for me, but I tried for a while to get 10+ working and it was completely impossible, 8 plus one of the simulation axes is the max I managed to get working. I changed a lot of code and finally pretty much hardcoded the descriptor for my use case, but, in case you are curious as to how I got slider and dial to work, here you have my descriptor.

` 
  buttonCount = 41; 
  uint8_t buttonPaddingBits = 7;

  uint8_t descriptor[] = {
		0x05, 0x01,                     // Usage Page (Generic Desktop Ctrls)
		0x09, 0x04,                     // Usage (Joystick)
		0xA1, 0x01,                     // Collection (Application)
		0x85, REPORT_ID,                //   Report ID (1)
		0x05, 0x09,                     //   Usage Page (Button)
		0x19, 0x01,                     //   Usage Minimum (0x01)
		0x29, _buttonCount,             //   Usage Maximum (buttonCount)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x25, 0x01,                     //   Logical Maximum (1)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x45, 0x01,                     //   Physical Maximum (1)
		0x75, 0x01,                     //   Report Size (1)
		0x95, _buttonCount,             //   Report Count (buttonCount)
		0x55, 0x00,                     //   Unit Exponent (0)
		0x65, 0x00,                     //   Unit (None)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
		0x75, 0x01,                     //   Report Size (1)
		0x95, buttonPaddingBits,        //   Report Count (buttonPaddingBits)
		0x81, 0x03,                     //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x30,                     //   Usage (X)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x31,                     //   Usage (Y)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x32,                     //   Usage (Z)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x33,                     //   Usage (Rx)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x34,                     //   Usage (Ry)
		0x15, 0x00,                     //   Logical Minimum (0)
[... 14 lines cut ...]
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x36,                     //   Usage (Slider)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0x05, 0x01,                     //   Usage Page (Generic Desktop Ctrls)
		0x09, 0x37,                     //   Usage (Dial)
		0x15, 0x00,                     //   Logical Minimum (0)
		0x27, 0xFF, 0x0F, 0x00, 0x00,   //   Logical Maximum (4095)
		0x35, 0x00,                     //   Physical Minimum (0)
		0x47, 0xFF, 0x0F, 0x00, 0x00,   //   Physical Maximum (4095)
		0x75, 0x10,                     //   Report Size (16)
		0x95, 0x01,                     //   Report Count (1)
		0x81, 0x02,                     //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)

		0xC0,                           // End Collection
	};`