schnoog/Joystick_ESP32S2 #26: Unable to make it work with ESP32 S3 Devkit

aljazjelen, 2025-06-16:
Hey! I am using the USB port with following config:

`[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
build_flags = 
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
;board_build.mcu = esp32s3
;board_build.variant = esp32s3
;monitor_speed = 115200
lib_deps = schnoog/Joystick_ESP32S2@^0.9.4`

and basic example (the one at the readme.md).

Device is continously detected as "USB Serial Device (COM 3)", regardless of what I do. 
And yes, I am using correct USB C port :)

aljazjelen, 2025-06-17:
Strange, when I set the flag
`-DARDUINO_USB_MODE=0`
it works. Well it seems so at least

schnoog, 2025-06-17:
Hi and thanks for your feedback. 
ARDUINO_USB_MODE isn't something I noticed before.
Espressif is shy on that topic: https://docs.espressif.com/projects/arduino-esp32/en/latest/search.html?q=ARDUINO_USB_MODE&check_keywords=yes&area=default

Will dig into it and update the documentation once I got some clarity on that topic.